openbmc/phosphor-pid-control #7: Use dbus active read for thermal sensors

mgersh, 2019-02-19:
Right now they're built as passive read sensors, so when the value doesn't change in a while, they go into failsafe mode.

feistjj, 2019-02-19:
If this changes it needs to be configurable. We rely on this so that if a sensor stops responding it can be treated as an error and properly boost the fans.

mgersh, 2019-02-19:
How are you handling the case of sensors not changing for a while? Long timeouts?

feistjj, 2019-02-19:
We're using dbus-sensors project not phosphor-hwmon, I'm not sure of the behavior of phosphor-hwmon but dbus-sensors keep an async handle on the file descriptor and update the value for any change, so it hasn't been an issue yet. We also don't have any host pushed sensors.

madscientist159, 2020-02-17:
We need this option as well.