openbmc/phosphor-pid-control #10: FR: separate tuning and logging

kunyi731, 2019-05-08:
One of the possible thermal tuning steps is having the inner PID loop + outer PID looping all running with configurations, and observe whether the thermal setpoint is oscillating too much. In order to complete the test, phosphor-pid-control needs to log while expecting a valid setpoint is supplied.

Current -t (TuningLogging) feature assumes the set point is hard coded, but it's probably more flexible to have options that enable tuning and logging separately.

pstrinkle, 2019-05-08:
When performing the tuning, typically a script is used that sets the set-point in the file to say, 6000, then after X seconds, 5000, then 7000, and tracks the fans abilities to get to that set-point -- whether they wobble too much.  Driving to a thermal set-point and seeing how the configuration performs can be done via setting a margin sensor (with a setpoint) via a dbus or ipmi command.

pstrinkle, 2019-05-08:
tuning vs logging separate -- easy.  If I break it out - yes.

commit de79ee05 (2019-05-08): split parameters for logging and tuning

add a flag to enable/disable tuning: default off
add an option for specifying a folder for outputting logs.

Closes: #10

Change-Id: I43864768f1dee8672f13288d3cf5a0c87c373aed
closed 2019-05-08 by bradbishop