openbmc/phosphor-pid-control #15: Why Fan do PID process again before output the pwm?

chineweff, 2019-08-19:
As I know, temps get the current temp value, and then Use temp as input do PID process to calculate new output pwm.
But after process calculates one output pwm, fans do PID process again before writing pwm to pwm file.

I don't understand why fans have to do pid again( use current fans output pwm as input)
Is anyone can explain about it? 
Tks

Awesomex005, 2021-09-03:
The FAN RPM will take time to catch up with the PWM setting. That is when you set 100% PWM, it takes some time for the fan from 0 RPM to reach 100% RPM.
The FAN PID is intent to have the Fan reach the desired speed (Thermal PID's output) fast and smooth.
Anyway, the FAN PID portion is optional, you may skip it base on your real design. Just set feedforward to 1 and PID coefficients to 0.

Krellan, 2021-09-25:
Good answer. Yes, there are 2 different PID loops. It's a pipeline:

The degrees of thermal safety margin remaining --> The necessary fan RPM to cool down the system

The desired fan RPM --> The PWM to command the fan motor to achieve that RPM

The reason for the second PID loop is that most fans don't let you set the RPM directly. Instead, all you can set is the PWM, from 0 to 100 percent of applied power to the fan motor.

Krellan, 2021-09-25:
Closing issue because question answered above.

closed 2021-09-25 by Krellan