openbmc/phosphor-pid-control #18: Cannot set dual-rotor with one PWM

alice-jabil, 2021-11-17:
I tried to config PID parameters with entity-manager config. I would like to define two rotor in input settings, as below ,But it will go fail.
Why PID only support 1 pwm to one tach in a fan zone?
How to support dual rotor to 1 pwm?

https://github.com/openbmc/phosphor-pid-control/blob/0001ee0224eea058e8bf0a71bd019e74f8b1738d/dbus/dbusconfiguration.cpp#L663

```
{
            "Class": "fan",
            "FFGainCoefficient": 5.0,
            "FFOffCoefficient": 0.0,
            "ICoefficient": 0.0,
            "ILimitMax": 0.0,
            "ILimitMin": 0.0,
            "Inputs": [
                "Fan_1_a",
                "Fan_1_b",
                "Fan_2_a",
                "Fan_2_b"
            ],
            "Name": "System_Fan",
            "NegativeHysteresis": 0.0,
            "OutLimitMax": 60.0,
            "OutLimitMin": 30.0,
            "Outputs": [
                "Pwm1",
                "Pwm2"
            ],
            "PCoefficient": 0.0,
            "PositiveHysteresis": 0.0,
            "SlewNeg": 0.0,
            "SlewPos": 0.0,
            "Type": "Pid",
            "Zones": [
                "Zone1"
            ]
        },

Krellan, 2021-12-03:
I suggest making two separate fan PID loops. This would let you have one PID loop per rotor. Each PID loop would manage the tach and PWM of a single rotor. I think this would give you the solution you are looking for.

alice-jabil, 2021-12-15:
Yes. Finally , I refer to [R1000.json](https://github.com/openbmc/entity-manager/blob/master/configurations/R1000%20Chassis.json), it helps me a lot .
Thanks for your help, Krellan!

closed 2021-12-15 by alice-jabil