openbmc/phosphor-pid-control #26: pidloop in phosphor-pid-control is running significantly slower than expected

gauravgandhi70, 2022-12-14:
Currently default values for cycleIntervalTimeMS and updateThermalsTimeMS  are 100 and 1000. These values provide the counter for how often ` processThermals` runs in [pidControlLoop](https://github.com/openbmc/phosphor-pid-control/blob/c612c051a5c414180f8ebe273c24543ab95d16ee/pid/pidloop.cpp#L49). getUpdateThermalsCycle gets [updated](https://github.com/openbmc/phosphor-pid-control/blob/c612c051a5c414180f8ebe273c24543ab95d16ee/pid/buildjson.cpp#L201) in `buildPIDsFromJson` function but this is not done in dbusconfiguration. Our platform does not use buildjson; it depends on dbusconfiguration method. Becasue of this by default pidControlLoop runs every 100ms and [cycleCnt](https://github.com/openbmc/phosphor-pid-control/blob/c612c051a5c414180f8ebe273c24543ab95d16ee/pid/pidloop.cpp#L115) gets increamented by 1 (every 100ms) and processThermal gets called when cycleCnt is 1000 causing 100 Sec delay.

CC: @Krellan

gauravgandhi70, 2022-12-15:
Original commit causing this issue: https://github.com/openbmc/phosphor-pid-control/commit/0e8fc3980a6690d1b2be2b4e96a8018aa1eacc96

@BonnieLo

Krellan, 2022-12-15:
I have an idea as to how to fix this. Let me see if it works, then I'll put it up for review.

harveyquta, 2022-12-15:
I commit a PR to implement cycle time setting in dbusconfiguration
https://gerrit.openbmc.org/c/openbmc/phosphor-pid-control/+/58797

Krellan, 2022-12-15:
Here's my intended fix: https://gerrit.openbmc.org/c/openbmc/phosphor-pid-control/+/59721

* Cleaned up D-Bus initialization path (to match the old static JSON initialization path)
* Redid the timer logic to remove the need for the division that was done earlier
* Also noticed and fixed a bug in timer logic that was causing it to run slower than intended

Krellan, 2022-12-15:
As for https://gerrit.openbmc.org/c/openbmc/phosphor-pid-control/+/58797

Good idea, I like how both of the settings paths were consolidated into a single function. I ended up having to do it slightly differently, as the old JSON path differs a little in the types of the variables from the new D-Bus path.

gauravgandhi70, 2022-12-16:
Thanks @Krellan , @harveyquta . Closing the issue!

closed 2022-12-16 by gauravgandhi70