openbmc/phosphor-pid-control #30: Sometimes return "Connection timed out" when set auto mode continuously

harveyquta, 2022-12-23:
When set auto mode continuously, it will sometimes return "Connection timed out". But when set to manual mode, it won't happen.

set auto mode cmd:
busctl set-property xyz.openbmc_project.State.FanCtrl /xyz/openbmc_project/settings/fanctrl/zone2 xyz.openbmc_project.Control.Mode Manual b false; \\
busctl set-property xyz.openbmc_project.State.FanCtrl /xyz/openbmc_project/settings/fanctrl/zone0 xyz.openbmc_project.Control.Mode Manual b false; \\
busctl set-property xyz.openbmc_project.State.FanCtrl /xyz/openbmc_project/settings/fanctrl/zone1 xyz.openbmc_project.Control.Mode Manual b false;

return:
Failed to set property Manual on interface xyz.openbmc_project.Control.Mode: Connection timed out

But if adding sleep between each busctl cmd, like below, the error will not happen.
busctl set-property xyz.openbmc_project.State.FanCtrl /xyz/openbmc_project/settings/fanctrl/zone2 xyz.openbmc_project.Control.Mode Manual b false; sleep 0.1; \\
busctl set-property xyz.openbmc_project.State.FanCtrl /xyz/openbmc_project/settings/fanctrl/zone0 xyz.openbmc_project.Control.Mode Manual b false; sleep 0.1; \\
busctl set-property xyz.openbmc_project.State.FanCtrl /xyz/openbmc_project/settings/fanctrl/zone1 xyz.openbmc_project.Control.Mode Manual b false;

harveyquta, 2022-12-23:
if set manual mode continuously, the error also will not happen.

Krellan, 2022-12-24:
Very strange. I wonder what would be causing this?

williamspatrick, 2022-12-24:
Daemon blocked on sysfs calls to update fan speeds and not able to respond to dbus fast enough?

harveyquta, 2023-01-06:
Seems like what @williamspatrick said.
After I mark the [dbus_call](https://github.com/openbmc/phosphor-pid-control/blob/master/dbus/dbuswrite.cpp#L80-L96) in dbus/dbuswrite.cpp, "Connection timed out" will not happen.
When set manual to false, [_redundantWrite](https://github.com/openbmc/phosphor-pid-control/blob/master/pid/zone.cpp#L89) will be set to true, and pid will [set property mandatorily one time](https://github.com/openbmc/phosphor-pid-control/blob/master/pid/fancontroller.cpp#L179).

Krellan, 2023-06-06:
This is a general problem, not just with the setting of automatic mode.

Sometimes D-Bus calls are just straight up forgotten about. The caller receives the timeout. The `swampd` program receives nothing at all, so it takes no action.

This problem happens rarely enough that it is tricky to debug. Any thoughts?