openbmc/phosphor-pid-control #36: [proposal] set failsafePwm by pid configuration

huangalang, 2023-05-10:
as tile 
we are sharing the design proposal about set failsafePwm by pid configuration

<motivation>
Current phosphor-pid-control service supports one zone with one failsafepwm 
However, for some platforms there are different plugable cards in a zone 
Different combinations may require different failsafe pwm settings
Why not just use one zone failsafepwm for all the combinations?
because if we take power consumption or acoustic into consideration
, we will find that not all the cases need the same high failsafepwm
Each case just need high enough failsafepwm to cool down the system in that condition 

For example
case1:  zone0 : gpuA card + motherboard in zone0
 =>requires failsafepwm = 80

case2:  zone1 : gpuB card + motherboard in zone0
 =>requires failsafepwm = 60

In order to solve the problem described above , we propose the design that
Each pidloop has its own failsafepwm
The final failsafepwm will be calculated during building zone process
The detailed design concept is described in design concept section

<design concept>
a zone has several pid loops, each one has its own pid failsafe pwm
defined in its configuration.

The calculation flow goes as bellow:
1. Take failsafepwm of the pidloops and the zone from the configuration files, if any is missing , set it to zero.
2. Take max value of (zone failsafepwm and pid loops failsafepwm).
3. If the max value from 'step 2' is zero indicates none of failsafepwm is configured, set it to default setting 100%.
[note]
1 this is intended for dbus configuration method , since one zone may have different pidloops from different plugable boards entity json , but this design also works for static configuration method
therefore , the zone with different boards will need different failsafepwm
2 . keep it back compatible with original design , also take zone failsafepwm into calculation
3 only take thermal controller pidloop into consideration

Working example as bellow

Case1 
Zone0:
gpuA pidloop failsafepwm = 70
Motherboard failsafepwm = 40

=>final failsafepwm = 70

Case2
Zone0:
gpuB pidloop failsafepwm = 80
Motherboard failsafepwm = 40

=>final failsafepwm = 70

blackcatevil, 2023-05-12:
> for some platforms there are different plugable cards in a zone Different combinations may require different failsafe pwm settings
In my opinion, I would say this could be done by the "Probe" function of entity manager to load the pid configuration dynamically if you can detect the type of device and put them on D-bus.

huangalang, 2023-06-09:
@blackcatevil 
the method is submitted , you can go and have a look