openbmc/phosphor-pid-control #41: Supports multiple fail-safe modes

averyzlyang, 2023-10-17:
The fan can have many different multiple failsafe values ​​(PWM) depending on different failsafe mode conditions
Example:
Case 1: When any fan fails, fan PWM is set to 100%.
Case 2: When some temperature sensors cannot be read, the fan PWM is set to 80%.
Case 3: Fan PWM set to 90% under other fault conditions

Krellan, 2023-10-31:
This is a good idea.

If we can enumerate all the desired failsafe conditions, that will help. All of these will be optional (user can configure them if necessary for their platform).

If more than one of these failsafe conditions is true at the same time, then a simple rule applies: take the maximum PWM of all those that are true, and use that maximum. This will disambiguate.

harveyquta, 2023-11-08:
Hi, I commit a PR to let failsafe duty more mutable. Does it meet your requirements?
https://gerrit.openbmc.org/c/openbmc/phosphor-pid-control/+/67566

Krellan, 2024-01-09:
This is a good idea, to allow each individual PID loop to set its own desired failure PWM percentage value. This would be optional, and if not set, the traditional zone-wide failure PWM value would still be applied. So, when a failure happens to any PID loop that doesn't have an individual PWM value set, the zone will still go into failsafe, using the zone-wide PWM value for failsafe.

However, this good idea does not really correspond with the initial comment from October 16 above. It is but one piece of the puzzle. More work will still be required, to implement the criteria in that comment from October 16 above.