openbmc/phosphor-pid-control #44: File Descriptor Leak Causing "Too Many Open Files" Error

fornchu110, 2025-02-20:
phosphor-pid-control service is experiencing a `System.Error.EMFILE: Too many open files` error due to a file descriptor leak.
This error occurs during the fan zone reconstruction process in `restartControlLoops`, resulting in the failure of fan zone creation and causing fan control operations to malfunction.

waffle2k, 2025-02-23:
We've locally patched this by removing the naive exception handling
```
diff --git a/main.cpp b/main.cpp
index e91d109..d821e6e 100644
--- a/main.cpp
+++ b/main.cpp
@@ -199,17 +199,8 @@ void tryRestartControlLoops(bool first)
             return;
         }
 
-        // retry when restartControlLoops() has some failure.
-        try
-        {
-            restartControlLoops();
-        }
-        catch (const std::exception& e)
-        {
-            std::cerr << "Failed during restartControlLoops, try again: "
-                      << e.what() << "\n";
-            tryRestartControlLoops(false);
-        }
+        // Allow exceptions to fall through.
+        restartControlLoops();
     };
 
     // first time of trying to restart the control loop without a delay
```
As the error condition will never resolve itself and the retry will be forever stuck. systemd will restart the service and unblock it, but this workaround doesn't solve the underlying FD leak.

williamspatrick, 2025-04-10:
Potential fix in Gerrit:

remote:   https://gerrit.openbmc.org/c/openbmc/phosphor-pid-control/+/79597 dbushelper: avoid new bus connections [NEW]

Krellan, 2025-04-10:
Nice, good find!

waffle2k, 2025-04-10:
> Looks good. Do you have this in a Gerrit patch, so that it can be merged?

I don't think that's necessary. Right now we're monitoring the number of coredumps being created on a per service basis, and `swampd` is a known issue for us while this gh issue is open. My plan is to let @williamspatrick 's gerrit commit make its way to our build repo, and when we eventually release a build with his fix we'll monitor for the occurrence of swampd aborting -- if it goes down to 0, we'll drop our downstream patch and continue to monitor.

closed 2025-05-09 by bradbishop